Webreaper/SolisAgileManager #174: Unexpected charge session

JamesM864, 2025-04-24:
At 5am this morning the app decided to send new charging instructions to the invertor and it charged the batteries between 5am and 5:30am. My rules do not specify this in any way. I charge up to 5am and only fill the batteries to 50%. at 5am the SOC was 49%.

[image]

You can see this in the log:
[05:00:01.751--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
[05:00:02.035--INF] Retrieved 3 rates from Octopus (24-Apr-2025 18:00 - 24-Apr-2025 15:00) for product E-1R-FLUX-IMPORT-23-02-14-G
[05:00:02.921--INF] Sending new charge instruction to Solis Inverter: 75, 0, 05:00-05:30, 00:00-00:00

I suspect it somehow got confused because I have the config "battery % for peak period" set to 50%.

[image]

I attach my config and the log file.

[SolisManagerConfig (1).json](https://github.com/user-attachments/files/19888561/SolisManagerConfig.1.json)
[SolisManager-20250424.log.bak.txt](https://github.com/user-attachments/files/19888560/SolisManager-20250424.log.bak.txt)

James.

Webreaper, 2025-04-24:
I think this is working as expected. The main goal of the algorithm is to ensure the battery is charged with the cheapest electricity, before the most expensive rate kicks in. So if you have "Battery % for Peak Period" set to 50%, then if your battery is below 50%, it'll try and charge to prepare for the peak period. 

Now, the 'peak period' is generally the highest price period for Agile, but the 'cheapest' period to charge in preparation for that depends on the prices before that peak. So it'll find the dip in the prices and charge then. 

For Flux, there is only one 'cheap' dip, from 2am to 5am. The rest of the day it's either 'average' or 'peak'. 

I suspect what happened was this (I'm basing this on the graph, I've not looked in detail at your graph):

1. At around 3am, you were at about 60% SOC
2. Your house consumed gradually over the next couple of hours
3. Just before 5am, when Flux is still in the 'cheapest' slot, your battery dipped under 50%, to 49%
4. This triggered the algorithm to try and charge the battery to achieve at least 50% SOC, during the cheapest time (which was immediately). 
5. Then 30 minutes later, the battery was over 50%, so the 'charge before peak' percentage had been reached, and the battery stopped charging.

So the app was doing exactly what was expected. You could argue that it was unlucky, because if the battery had dropped below 50% about 5 minutes later, it wouldn't have charged. 

Either way, though, the solution is to reduce the 'Battery % for the peak' setting - perhaps drop it to 40%, or even less, to ensure that it really only charges during that 2am-5am period if the battery is very low.

Hope that makes sense.

JamesM864, 2025-04-24:
Fair enough. I don't think I will use this charge before peak option. I prefer to have more control over when charging happens!

James.

Webreaper, 2025-04-24:
Yeah, sorry - that's really the underlying reason and algorithm for the app, so it doesn't really make sense for me to disable that.

TBH, if you want more manual control over when charging happens, then I'm not sure the app will help, because the whole point behind it is that you just leave it to 'do its thing' and it optimises the charging for you.  If you really just want to use the app as a way to manage charging without going through Soliscloud, then your best solution would be to set charge-before-peak to something low, like 20%, and then just use scheduled actions, or something like that.

Webreaper, 2025-04-24:
Yeah, the "never charging when it's not needed" is more complex than you think. And generally, from what I've found so far since starting to write the app, the more important thing is to not miss charging when you **do** need it. 

If the battery is charged and it turns out to be a good solar day, then it's not the end of the world as export usually offsets it (and in the case of this ticket, if you're charging at 16p, and exporting at 15p, it all nets out). 

Whereas if it'd a dull rainy day, and the battery doesn't have enough charge to get you through the 40p/kwh peak while you're cooking etc, then it'll cost you real money.

But like all these things, it's tweaks and adjustments....

closed 2025-04-24 by Webreaper