Webreaper/SolisAgileManager #180: No IOG Dispatch info being received from Octopus

dqj999, 2025-04-30:
When I charged my car on 29th April, running into 30th April, the logs show that despite checking for IOG Smart Charge slots no dispatch information was received. This was despite my car being Smart Charged for several hours.

The last time that I charged my car on 24th April the charging sessions were correctly reported.

[image]

[image]

[SolisManager-20250429.log](https://github.com/user-attachments/files/19975201/SolisManager-20250429.log)
[SolisManager-20250430.log](https://github.com/user-attachments/files/19975202/SolisManager-20250430.log)

Webreaper, 2025-04-30:
Please raise this issue with Octopus. As discussed several times before, there is literally nothing I can do to fix this - if I make the call to check the IOG dispatches, and Octopus doesn't return one, then I can't do anything about it.

closed 2025-04-30 by Webreaper

dqj999, 2025-04-30:
Yes fair enough - should have put this as a discussion. I've posted something on the Octopus Energy Reddit group about this. I'll also raise it with an email to Octopus.

Webreaper, 2025-04-30:
Yeah, reddit forum and an email to Octopus is the best answer.