Webreaper/SolisAgileManager #182: Origin of Actual PV (kWh) data

By-tor, 2025-05-02:
Hi guys, where does the data originate for the Actual PV (kWh) in the history page?

I have just notice that one of my sites is, and always has been, reporting zero Actual PV (This would explain why I never got negative values too)

I suspect it's a Solis issue and nothing to do with this app.

[image]

Webreaper, 2025-05-02:
It comes from the inverter. The log entries that say "Enriching history with PV yield" are when the app is querying the inverter to back-fill the history with the PV and other stats. 

It shouldn't be coming back as zero though. Are you running the latest version?

By-tor, 2025-05-02:
Yes, I'm running the latest version.

I have just noticed though that the Solis Cloud app isn't reporting any solar generation apart from the instantaneous values.
For me this proves it's either a setting in the Solis Cloud app or a Solis Issue.

I'll raise a ticket with Solis.

[image]

Webreaper, 2025-05-02:
Good luck!

closed 2025-05-06 by Webreaper