Webreaper/SolisAgileManager #183: No Costs Data

By-tor, 2025-05-02:
On the Costs page (v1.0.643.0) I am not getting any data, just shows the loading screen...

[image]

Am I too impatient - 20 mins and counting?

Webreaper, 2025-05-02:
No, probably means that there's an error, or that the code to pick out the right meters is falling. Any errors or messages in the log?

For me, it takes about 5s to load the 30 day costs, and maybe 20s to load 6 months.

Webreaper, 2025-05-02:
But also, I rewrote a bunch of that logic last night so try the very latest build, it might be better.

By-tor, 2025-05-07:
No, I have 1.0.655 running and there are a few changes but still no data

Webreaper, 2025-05-07:
Can you send me the snippet of the logs from when it calculates the costs? E.g., for me, I see:
```
[09:32:38.854--INF] Querying consumption data from "2025-04-05T08:32:38.4050000Z" to "2025-05-05T08:32:38.4050000Z"
[09:32:38.854--INF]   Requesting consumption data for import meter S/N [REDACTED]
[09:32:38.855--INF]   Requesting consumption data for export meter S/N [REDACTED]
[09:32:39.710--INF]   Found 1 tariffs for meter. Querying rates (Tariffs: E-1R-COSY-22-12-08-J
[09:32:39.872--INF] Retrieved 211 rates from Octopus (05-Apr-2025 06:00 - 05-May-2025 12:00) for product E-1R-COSY-22-12-08-J
[09:32:39.876--INF]   Found 1 tariffs for meter. Querying rates (Tariffs: E-1R-OUTGOING-VAR-24-10-26-J
[09:32:39.958--INF] Retrieved 1 rates from Octopus (25-Oct-2024 23:00 - today) for product E-1R-OUTGOING-VAR-24-10-26-J
```
Can you check if the meter serial number for the import/export meters are the correct ones  It _should_ be the same one listed on your Octopus bill, where it says "Energy Charges for Meter [REDACTED]".

By-tor, 2025-05-07:
We appear to have found the problem:

[12:20:37.949--INF] Querying consumption data from "2025-04-05T08:24:35.2120000Z" to "2025-05-05T08:24:35.2120000Z"
[12:20:37.949--WRN] Attempted to get consumption, but no account number specified

Webreaper, 2025-05-07:
Ah, yes, it won't work if you don't specify an Octopus Account number and API key in the config/settings. 

That said - the 'Costs' screen should display a warning for that... did it not?

Webreaper, 2025-05-08:
Don't really understand how the app can work at all if the config isn't being picked up correctly, but.... 

I've added some additional logging to the latest dev build - could you deploy it and send me the startup logs?

By-tor, 2025-05-08:
Hi, not sure I have done this correctly but results still appear the same.
I couldn't see any later dev version so I pulled 1.0.655.0  - Docker said that was up to date - so maybe I'm not pulling the correct version. Sorry I don't use git that much

Re-ren app and grabbed the log files:-

[20:26:44.850--INF] ===========================================================
[20:26:44.883--INF] Application started. Build version v1.0.655.0 Logs being written to /appdata
[20:26:44.884--INF] Running in Docker
[20:26:44.884--INF] Current timezone: BST
[20:26:45.039--INF] Loaded cached Solcast data from /appdata/Solcast-cache.json
[20:26:45.233--INF] Now listening on: http://[::]:5169
[20:26:45.235--INF] Application started. Press Ctrl+C to shut down.
[20:26:45.235--INF] Hosting environment: Production
[20:26:45.235--INF] Content root path: /app
[20:26:45.846--INF] Found Octopus Product/Contract: E-1R-AGILE-24-10-01-D, Starts 15-Feb-2025
[20:26:45.854--INF] Battery SOC is zero on first refresh - forcing update from inverter...
[20:26:46.048--INF] Refreshed state: SOC = 54%, Current PV = 0.10kW, House Load = 0.82kW, Forecast today: 17.13kWh, tomorrow: 22.82kWh
[20:26:46.310--INF] Retrieved 54 rates from Octopus (08-May-2025 19:00 - 09-May-2025 22:00) for product E-1R-AGILE-24-10-01-D
[20:26:46.314--INF] 54 new Octopus rates available to 09-May-2025 23:00 (cheapest: 14.553p/kWh, peak: 36.015p/kWh)
[20:26:46.348--INF] Loaded 4143 entries from execution history file SolisManagerExecutionHistory.csv
[20:26:46.422--INF] Forecast = 22.82kWh (so > 9.0kWh). Found 0 overnight charge slots to skip between "2025-05-08T20:30:00.0000000+01:00" => "2025-05-09T07:00:00.0000000+01:00"
[20:26:47.281--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
[20:26:47.295--INF] Enriching history with PV yield for 5 days
[20:26:47.669--INF] Retrieved 233 inverter stats for 08-May-2025
[20:26:47.977--INF] Retrieved 288 inverter stats for 04-May-2025
[20:26:48.291--INF] Retrieved 287 inverter stats for 01-May-2025
[20:26:48.590--INF] Retrieved 289 inverter stats for 30-Apr-2025
[20:26:48.894--INF] Retrieved 287 inverter stats for 06-Mar-2025
[20:26:49.211--INF] Updating inverter time to 2025-05-08 20:26:49 avoid drift...
[20:28:00.406--INF] Refreshed state: SOC = 53%, Current PV = 0.00kW, House Load = 1.51kW, Forecast today: 17.13kWh, tomorrow: 22.82kWh
[20:30:01.722--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
[20:30:02.200--INF] Retrieved 53 rates from Octopus (08-May-2025 19:30 - 09-May-2025 22:00) for product E-1R-AGILE-24-10-01-D
[20:30:02.203--INF] Forecast = 22.82kWh (so > 9.0kWh). Found 0 overnight charge slots to skip between "2025-05-08T20:30:00.0000000+01:00" => "2025-05-09T07:00:00.0000000+01:00"
[20:30:02.705--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
[20:30:02.711--INF] Enriching history with PV yield for 5 days
[20:30:03.127--INF] Retrieved 234 inverter stats for 08-May-2025
[20:31:02.976--INF] Querying consumption data from "2025-04-07T19:31:02.8050000Z" to "2025-05-07T19:31:02.8050000Z"
[20:31:02.977--WRN] Attempted to get consumption, but no account number specified

Webreaper, 2025-05-08:
Are you using docker or direct install? If the latter, you want the
binaries from this build.

https://github.com/Webreaper/SolisAgileManager/actions/runs/14901407312

By-tor, 2025-05-09:
I have been using Docker - it's the only way I am familiar with !!
Using :-

docker pull webreaper/solisagilemanager:1.0.655

Not sure how to get the dev versions (apologies for ineptitude)

Webreaper, 2025-05-09:
Ah, I see. If you change your docker run command (or docker-compose) to use `webreaper/solisagilemanager:dev` then you'll get the latest dev build whenever you pull and restart. 

When you're not using the dev version (it can be unstable sometimes) It's better to use `webreaper/solisagilemanager` (or add `:latest` to ensure you're not pinned to a specific version and always pick up the latest release whenever you upgrade/pull.

By-tor, 2025-05-09:
Ok, so got the correct version (1.0.660) now and this is what I get:-

[17:04:19.578--INF] ===========================================================
[17:04:19.611--INF] Application started. Build version v1.0.660.0 Logs being written to /appdata
[17:04:19.612--INF] Running in Docker
[17:04:19.612--INF] Current timezone: BST
[17:04:19.660--INF] Config loaded
[17:04:19.660--INF]   Octopus Account number: A-E3xxxxxx
[17:04:19.660--INF]   Octopus Product Code: E-1R-AGILE-24-10-01-D
[17:04:19.660--INF]   Octopus API Key was specified
[17:04:19.660--INF]   Scheduled actions configured: 0
[17:04:19.772--INF] Loaded cached Solcast data from /appdata/Solcast-cache.json
[17:04:19.967--INF] Now listening on: http://[::]:5169
[17:04:19.969--INF] Application started. Press Ctrl+C to shut down.
[17:04:19.969--INF] Hosting environment: Production
[17:04:19.970--INF] Content root path: /app
[17:04:19.988--INF] Battery SOC is zero on first refresh - forcing update from inverter...
[17:04:20.353--INF] Refreshed state: SOC = 100%, Current PV = 1.50kW, House Load = 0.93kW, Forecast today: 22.89kWh, tomorrow: 21.84kWh
[17:04:20.632--INF] Retrieved 60 rates from Octopus (09-May-2025 16:00 - 10-May-2025 22:00) for product E-1R-AGILE-24-10-01-D
[17:04:20.638--INF] 60 new Octopus rates available to 10-May-2025 23:00 (cheapest: 2.31p/kWh, peak: 36.015p/kWh)
[17:04:20.674--INF] Loaded 4185 entries from execution history file SolisManagerExecutionHistory.csv
[17:04:20.739--INF] Forecast = 21.84kWh (so > 9.0kWh). Found 0 overnight charge slots to skip between "2025-05-09T20:30:00.0000000+01:00" => "2025-05-10T07:00:00.0000000+01:00"
[17:04:21.428--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
[17:04:21.440--INF] Enriching history with PV yield for 5 days
[17:04:21.782--INF] Retrieved 193 inverter stats for 09-May-2025
[17:04:22.207--INF] Retrieved 288 inverter stats for 04-May-2025
[17:04:22.499--INF] Retrieved 287 inverter stats for 01-May-2025
[17:04:22.782--INF] Retrieved 289 inverter stats for 30-Apr-2025
[17:04:23.061--INF] Retrieved 287 inverter stats for 06-Mar-2025
[17:04:23.700--INF] Found Octopus Product/Contract: E-1R-AGILE-24-10-01-D, Starts 15-Feb-2025
[17:04:23.703--INF] Updating inverter time to 2025-05-09 17:04:23 avoid drift...
[17:05:01.417--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
[17:05:30.056--INF] Querying consumption data from "2025-04-08T16:05:30.0531845Z" to "2025-05-08T16:05:30.0531845Z"
[17:05:30.057--WRN] Attempted to get consumption, but data was returned

Webreaper, 2025-05-09:
Looks fine. I presume you redacted your account number with xxxxx?

I genuinely don't know what's wrong then or how you could get an error when you try and load the costs screen. 😕

Webreaper, 2025-05-09:
Oh, wait the last message is different this time. Let me look into that.

Webreaper, 2025-05-09:
Okay, had a look. I think it's failing to get the current meter data from your account. I've pushed a new version which is building and should be ready in a few minutes which will narrow this down.

We'll get to the bottom of it!

By-tor, 2025-05-09:
Latest version:

BTW - I don't have an export account or export meter !!

[18:31:31.525--INF] ===========================================================
[18:31:31.584--INF] Application started. Build version v1.0.661.0 Logs being written to /appdata
[18:31:31.585--INF] Running in Docker
[18:31:31.585--INF] Current timezone: BST
[18:31:31.641--INF] Config loaded
[18:31:31.641--INF]   Octopus Account number was specified
[18:31:31.646--INF]   Octopus API Key was specified
[18:31:31.646--INF]   Octopus Product Code: E-1R-AGILE-24-10-01-D
[18:31:31.646--INF]   Scheduled actions configured: 0
[18:31:31.785--INF] Loaded cached Solcast data from /appdata/Solcast-cache.json
[18:31:31.976--INF] Now listening on: http://[::]:5169
[18:31:31.981--INF] Application started. Press Ctrl+C to shut down.
[18:31:31.982--INF] Hosting environment: Production
[18:31:31.982--INF] Content root path: /app
[18:31:32.008--INF] Updating inverter time to 2025-05-09 18:31:32 avoid drift...
[18:31:33.967--INF] Battery SOC is zero on first refresh - forcing update from inverter...
[18:31:34.085--INF] Refreshed state: SOC = 94%, Current PV = 0.30kW, House Load = 0.76kW, Forecast today: 22.89kWh, tomorrow: 21.84kWh
[18:31:34.350--INF] Retrieved 57 rates from Octopus (09-May-2025 17:30 - 10-May-2025 22:00) for product E-1R-AGILE-24-10-01-D
[18:31:34.356--INF] 57 new Octopus rates available to 10-May-2025 23:00 (cheapest: 2.31p/kWh, peak: 36.015p/kWh)
[18:31:34.393--INF] Loaded 4188 entries from execution history file SolisManagerExecutionHistory.csv
[18:31:34.463--INF] Forecast = 21.84kWh (so > 9.0kWh). Found 0 overnight charge slots to skip between "2025-05-09T20:30:00.0000000+01:00" => "2025-05-10T07:00:00.0000000+01:00"
[18:31:35.280--INF] Inverter already in correct state (0, 0, 00:00-00:00, 00:00-00:00) so no charge instructions need to be applied
[18:31:35.296--INF] Enriching history with PV yield for 5 days
[18:31:35.665--INF] Retrieved 210 inverter stats for 09-May-2025
[18:31:35.968--INF] Retrieved 288 inverter stats for 04-May-2025
[18:31:36.293--INF] Retrieved 287 inverter stats for 01-May-2025
[18:31:36.583--INF] Retrieved 289 inverter stats for 30-Apr-2025
[18:31:36.884--INF] Retrieved 287 inverter stats for 06-Mar-2025
[18:31:37.488--INF] Found Octopus Product/Contract: E-1R-AGILE-24-10-01-D, Starts 15-Feb-2025
[18:31:59.435--INF] Querying consumption data from "2025-04-08T17:31:59.4130000Z" to "2025-05-08T17:31:59.4130000Z"
[18:31:59.437--WRN] No export meter found in account
[18:31:59.437--WRN] Attempted to get consumption, but no data was returned

Webreaper, 2025-05-09:
Ah. Do you have export set up with Octopus? Maybe you're import only?

By-tor, 2025-05-09:
No, I don't export to Octopus - as far as they are concerned I am import only
I have a roof rental arrangement with the PV supplier from 10+ years ago - I get nothing.

To further complicate matters I am using your app on my brothers system - his is behaving similarly to mine but I think for a different reason.
He has Import and Export with Octopus but I think he has different accounts with Octopus for Import and Export so the app shows no data for his costs page either

Webreaper, 2025-05-09:
Ah, okay, that makes sense.

So I can change it so it calculates import, and ignores export if you don't have an export meter.

For your brother, maybe I'll add config to specify an export account.

Webreaper, 2025-05-09:
Okay, I've pushed a change so that pulling in export data will be optional. You'll just see import costs, without any export offset. Give it about 10 mins and pull the `dev` tag and hopefully costs should work for you!

I'll add a separate ticket to support multiple Octopus accounts with separate import/export as a future enhancement.

By-tor, 2025-05-10:
Success !! I have import data - thank you

Webreaper, 2025-05-10:
Yeah, could be tricky. If he has 2 accounts, I'm unsure if each will need its own API key...

Might be difficult to debug without having 2 accounts. Could you send me his API key(s) and account number (to mark@otway.com) so I can see if it's possible? Only if you trust me, of course. 😁

By-tor, 2025-05-10:
Details emailed

Webreaper, 2025-05-10:
I've not received anything (checked spam etc too). Could you retry?

By-tor, 2025-05-10:
strange - it only just sent when I went into outlook

Webreaper, 2025-05-10:
Got it. Will take a look.

Webreaper, 2025-05-11:
Working!! 

[image]

Webreaper, 2025-05-11:
Just pushed the changes: https://github.com/Webreaper/SolisAgileManager/actions/runs/14959865068

You need to enter the two account numbers, comma-separated, in to the accounts settings field in the config. The order doesn't matter.

By-tor, 2025-05-11:
Excellent! will give it a test tomorrow! thanks again.

By-tor, 2025-05-12:
Works great. Thanks @Webreaper

closed 2025-05-12 by Webreaper