Webreaper/SolisAgileManager #185: Charging when shouldn't

By-tor, 2025-05-12:
Last night my batteries charged between 03:30 and 04:00 when set to not charge overnight if solar forecast for the day is above 9 kWh.

 As per the log:-

 At 00:03 Forecast today: 20.72kWh, tomorrow: 18.96kWh

At 01:00 Forecast today: 18.96kWh, tomorrow: 4.67kWh

At 03:30 Sending new charge instruction to Solis Inverter: 60, 0, 03:30-04:00, 00:00-00:00

At 09:13 Forecast today: 19.99kWh, tomorrow: 22.34kWh (after Solcast update)

Is the app using the wrong forecast figure (tomorrow's instead of today's - i.e. 4.67kWh) or is this something odd about Solcast?
Also, why does tomorrow become today at 01:00?
And where did the 4.67kWh go? How did it get to change to 22.34kWh?

[SolisManager-20250512.log](https://github.com/user-attachments/files/20161122/SolisManager-20250512.log)

Webreaper, 2025-05-12:
At 00:03, the 20.72kwh forecast will be for the day that's just ending. Then it rolls over to the next day, which will be 18.96kW. 

Tomorrow becomes today at 01:00 to avoid DST problems with the rollover at midnight. 

As for why it's using the 4.67kWh figure I'm not sure, it should have skipped the charge. However, I don't see a solcast update at 19:13 (which it uses to get the next 12 hours' forecast to check the NOC settings - which means it may have had an incomplete forecast perhaps? Let's see if it does better tonight.

By-tor, 2025-05-12:
Just checked recent logs and it seems the 19:13 call to Solcast always fails 

[image]

Webreaper, 2025-05-12:
Interesting. Are you sharing a single solcast account for multiple usages - perhaps in your own HA installation? 

Solcast is pretty fierce at rate-limiting. They only allow you 10 API calls every 24 hours on the free tier. If you have 2 strings, SolisManager needs to make 2 calls to get the forecasts - so if you have the extra mid-day call enabled in the settings, that'll be 6 API calls per day. If you _also_ have HA running etc., also querying Solcast, then you'll blow through your 10 very quickly. 

I'd probably recommend disabling the 'Extra Solcast Update' to free up a couple of calls. If you have multiple strings, I think you might be able to set up 2 accounts, one for each string, but unsure (@harryflatter may be able to opine, I know he solved this). 

If you're using the same account for your own setup and your brother's installation, that's likely to be the issue. I'd recommend registering for an independent Solcast account for each property.

It's also weird that you have one entry hitting Solcast at 09:04, but I presume that was just bad timing when you restarted the app?

By-tor, 2025-05-12:
I already have separate Solcast accounts for each installation - I will have a look at HA setup.

By-tor, 2025-05-12:
Worked fine at 19:13

Webreaper, 2025-05-12:
The 09:04 one should have read in the cached copy of the data - I'll look at tweaking that. It shouldn't use up all your API calls just because you restarted....

By-tor, 2025-05-12:
HA was using all the API calls during the day - every hour between 9am and 7pm is it? So the app used 1 call at 9:13 and there were none left by 19:13
I would be more concerned as to why the app used the 4.67kWh figure - doesn't Solcast provide 3 days data in advance?
I'd be surprised if their forecast jumped from 4.67 to 19.86

Webreaper, 2025-05-12:
I create a list of 30-minute forecasts from Solcast, which gets appended and extended when each successful call is made. I then total up all the forecasts in the 24-hour period for each day.

My hunch is that if the API calls failed, the number of 30-minute slots with a non-zero forecast became low as the future forecasts 'ran out', so maybe there was only 4-6 hours of forecasts, which only totalled 5kWh. 

I've had no problems at all with inconsistent readings and I've not had reports from anyone else (and I know quite a few are using it). Let's see how it works now there's some API calls for it to use.

By-tor, 2025-06-17:
Had another unexpected charge event this morning at 5am (so after the no charge at night)
SOC was 44% so above my charge below 10% threshold but it still charged after it had an error reading control state from the inverter. 
Should this failsafe to not charging if it has an error?

[SolisManager-20250617.log](https://github.com/user-attachments/files/20783077/SolisManager-20250617.log)

By-tor, 2025-06-19:
Would it be worth keeping the result of the previous successful inverter state returned and falling back to use that if the current API call fails?

Webreaper, 2025-06-19:
It may already do that, but I'd need to check. The problem, though, is that if you're using 30-minute stale data, the charging decision will be wrong in a different way. :)

Let me investigate more and see...