Webreaper/SolisAgileManager #186: Erroneous positive PV Yield values

By-tor, 2025-05-13:
Had a couple of strange values for PV Yield at 00:00 and 01:00 last night - only on one of my installations though which are both using 1.0.664
The two values are 10.24 which strangely is the sum of the previous day's yield
Anyone else had a similar issue? 
Could it have anything to do with this particular installation using two different Octopus accounts for import and export?

[image]

[SolisManager-20250513A.log](https://github.com/user-attachments/files/20185467/SolisManager-20250513A.log)

dqj999, 2025-05-14:
This sounds like it is related to the earlier bug with negative values for yield etc. Can’t the app just use zero values during the hours of darkness, whatever the source data is saying.

Webreaper, 2025-05-14:
Could do. I'll have a look at some point.