Webreaper/SolisAgileManager #189: Charge slots need enabling?

tabannis, 2025-05-26:
I've had my inverter firmware updated today. I'm not sure if this is new because of that or not, but I now have to "enable" my charge slots.

[image]

I noticed this because the 30 minutes app slot was set to charge, the logs said the inverter was in the right state, but wasn't charging. I had to "enable" the 1st charge slot via the official solis app.

Webreaper, 2025-06-22:
Did you discharge using the Soliscloud app, or SolisManager?

The rules for automatically discharging are that it'll discharge if there's enough zero/negative pricing to discharge the battery fully and then fully recharge it for free. Let me know if that was the case and it didn't work. I'm not on Agile right now so won't be able to test myself.

tabannis, 2025-06-22:
The app 👍 The discharge slot was set correctly but as I hadn't enabled it, the inverter ignored it

Webreaper, 2025-06-22:
Oh, I see what you mean. Yes, I need to look at making the API enable the slot itself. Probably need my firmware upgraded to test!

tabannis, 2025-06-22:
> Oh, I see what you mean. Yes, I need to look at making the API enable the slot itself. Probably need my firmware upgraded to test! 

I'd be happy to help if you'd rather not risk your inverter going odd, but Solis was quite swift and helpful at updating everything remotely and I've had no issues (apart from the EMS app functionality that was there, and now isn't 🤷‍♂️)

Webreaper, 2025-08-05:
Pushed to dev - if somebody could: 

1. Stop SolisManager
2. Disable the slots via the SolisCloud App
3. Run up the latest build
4. Check that when the first charge is applied, the charge / disacharge for slot 1 are both automatically enabled

That'd be great.

tabannis, 2025-08-05:
Tired that.in dev, and my charge slot did not get enabled.

I'll add the log shortly...

Webreaper, 2025-08-06:
Dammit. Will dig further into the Solis API....

Webreaper, 2025-08-07:
Sorry, forgot to press the "build" button. #701 is building now..... 😁

tabannis, 2025-08-08:
Ok, New logs here....

[SolisManager-20250808.log](https://github.com/user-attachments/files/21693117/SolisManager-20250808.log)

By-tor, 2025-08-28:
Am I reading the logs correctly - this is not fixed? 
I am holding off updating the firmware.

Webreaper, 2025-08-28:
It's not fixed yet, I haven't had a chance to look into it. I probably need to get my firmware upgraded so it's easier to test.

@By-tor you don't need to hold off updating; this issue is a minor inconvenience - you just go into the inverter control screen and check the checkbox to enable the slot. Once you've done it once, SolisManager will work just fine. The fix for this issue is just for me to adjust SolisManager so it checks that box automatically.