Webreaper/SolisAgileManager #199: Minor mobile display issue

abrasive-blueberry, 2025-07-05:
Just a minor display here since the update to show the sun and moon icons - on mobile in portrait orientation, the main screen no longer quite fits the screen and has to be scrolled (a tiny bit) horizontally.

I’ve only tested this on my iPhone 14 with the safari browser. It displays ok on larger screens like my iPad.

I notice that you posted a screenshot the other day with this layout and it looked ok, so it may be phone model or browser specific.

[image]

abrasive-blueberry, 2025-07-07:
Thanks. I shall test it out this evening and get back to you

abrasive-blueberry, 2025-07-07:
At the risk of making myself look foolish, I'm not entirely sure how to install the dev version. I've always just used wget to download the zip files containing the releases.

I can see the files that you have changed, but the dev file structure seems to be different from the file structure contained in the zip files, and I'm not 100% sure what to do with them.

Sorry, I'm guessing I maybe need to figure out how to use git.

Webreaper, 2025-07-07:
Just download the binaries you need from here:

https://github.com/Webreaper/SolisAgileManager/actions/runs/16111357528

abrasive-blueberry, 2025-07-08:
I know I’m going to embarrass myself here, but I’ve been through everything on that page and can’t find the binaries to download. Can I email you about this?

abrasive-blueberry, 2025-07-08:
Sorry for being so dim. I got there in the end and can confirm that this build has resolved the issue. The sun/moon are now hidden on my phone screen, but show up ok on my laptop and iPad.