Webreaper/SolisAgileManager #202: Option to prevent discharging

By-tor, 2025-08-05:
Could you add a settings option to prevent battery discharge please.
I don't get any income from any exported energy and discharge and re-charge cycles are just adding to the wear on my batteries reducing their service life for no benefit.

Webreaper, 2025-08-05:
Do you mean for when prices go negative? Even if you don't earn export, if prices are negative and you discharge and charge your battery, you'll make money because they're paying you to charge it.

But yes, I can add an option to never automatically discharge. Did the discharge kick in yesterday with Agile's negative pricing?

Webreaper, 2025-08-05:
I'll just put a 'never automatically discharge' flag in for now, as I don't have time for anything more complex. I'll see if I can get it done this week (which guarantees that a) prices will go negative until I ad the flag and b) prices won't go negative for 6 months after that.....).

By-tor, 2025-08-28:
Apologies, only just noticed you made the change.
Have deployed it now and looks ok. Will report back any issues, probably need to wait till the next negative price to prove it.

By-tor, 2025-09-01:
Negative prices showed up over the weekend !!

Both of my systems on dev v1.0.702.0

System A - set to disable auto discharge  - did not auto discharge
System B - set to enable auto discharge  - did not auto discharge either

Do you need logs to investigate if this was correct?

Webreaper, 2025-09-01:
I'll take a look - can you attach logs for both (or send them to me at mark@otway.com)

By-tor, 2025-09-01:
Sent to your email