Webreaper/SolisAgileManager #214: Intelligent Go Charging Checkbox

iang66, 2025-09-22:
Hi, 

Installed through Docker on Unraid and I'm trying to complete setup.

Am on Intelligent Go from Octopus and I'm looking for the "Intelligent Go Checkbox" in the settings screen that is referenced in the documentation.

Have set API from Octopus, Solis & Solcast successfully and info is being displayed correctly but cannot find the checkbox.

Any guidance gratefully received

MartyMoose30053, 2025-09-22:
I noticed this, you have to hit save and test charge, it then appeared

iang66, 2025-09-22:
Thank you both, screenshot taken and ready to send, test charge initiated.. will update shortly!

iang66, 2025-09-22:
Test charge unsuccessful, rebooted and tried again, no joy.

IOG checkbox not there, tariff showing in config logs as "Found Octopus Product/Contract: E-1R-INTELLI-FIX-12M-25-04-10-B"