Webreaper/SolisAgileManager #216: Webpage Corrupted Display

Catnip2007, 2025-09-30:
I've previously had the app working perfectly, but after restarting RPi4 the homepage on mobile is correct, but the pc webpage is corrupted, it just displays very large icons for Home, History, Comparison etc. At the bottom of the page is the message:

Initialising...
An unhandled error has occurred. [Reload]

I've tried restarting Rpi4, closing/opening webpage, rebooting pc etc and it's still the same.
I'm sure it's a simple issue, I searched the Issues without success.....

[image]

Webreaper, 2025-10-01:
> do I then need to disable the function to get it to be active?

No, the NOC function needs to be enabled to be active. I think you may be misunderstanding what NOC is for - it's to skip charging the battery the night before if you're forecasted to generate a lot the next day.

In the log above, Your threshold for No-Overnight charge is set to 12kWh. That means that if the forecast is more than 12kWh of solar generation, the battery will not be charged overnight (because the assumption is that you'll generate enough solar through the day to charge the battery before the peak period).

The screenshot shows that the forecast is expecting you to generate just 3.33kWh of energy via Solar. So the "No overnight charge" rule will not trigger, so the battery **will** be charged overnight.

Catnip2007, 2025-10-01:
Thanks again for clarifying the NOC function.
I think I've set it up correctly but don't see any scheduled charge activity, do I need to set charge current or timings somewhere?

[image]

[image]